Frix-x/klippain #472: Erreur Klipper

Didou30, 2024-02-13:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-14-g32344ff2

### Describe the bug and expected behavior

rapport de Klipper: ERROR

Section 'gcode_shell_command system_info' is not a valid config section

### Additional information and klippy.log

Config error
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/biqu/klipper/klippy/klippy.py", line 145, in _read_config
    pconfig.check_unused_options(config)
  File "/home/biqu/klipper/klippy/configfile.py", line 302, in check_unused_options
    raise error("Section '%s' is not a valid config section"
configparser.Error: Section 'gcode_shell_command system_info' is not a valid config section
webhooks client 281473142293024: New connection
webhooks client 281473142293024: Client info {'program': 'Moonraker', 'version': 'v0.8.0-310-g67c98f6'}

Frix-x, 2024-02-13:
This happens when the Gcode shell command extension is not installed. Please install it using KIAUH (in advanced section) and it should work :)

closed 2024-02-13 by Frix-x

Didou30, 2024-02-13:
je n'utilise pas KIAUH 5btt py) mais j'ai fait une réinstallation de Klippain depuis le SSH et cela fonctionne merci