Frix-x/klippain #473: LDO_Nighthawk-SB Toolhead board config.

mvdveer, 2024-02-13:
### Describe the feature or hardware support you'd like

Support for LDO Nighthawk-SB board. I have done a mapping of the pins and included it in the attached config file as a .pdf as it won't upload as a .cfg file.  I was going to add the file to the  /config/mcu_definitions/toolhead section and reference it in the mcu.cfg file [include config/mcu_definitions/toolhead//LDO_Nighthawk_v1.0.cfg]

### Additional context or information

[LDO_Nighthawk-SB_v1.0.pdf](https://github.com/Frix-x/klippain/files/14272616/LDO_Nighthawk-SB_v1.0.pdf)

mvdveer, 2024-02-13:
Link to LDO pin Mappings:
[https://docs.ldomotors.com/en/voron/nitehawk-sb](url)

Phantastic1, 2024-02-14:
following... I'm currently running the LDO Nighthawk and would like to see a configuration for this setup too

Surion79, 2024-02-14:
you are always welcome to take a config as template and provide a configuration via a pull request to speed up the process :)

mvdveer, 2024-02-14:
> you are always welcome to take a config as template and provide a configuration via a pull request to speed up the process :)

I uploaded it as a .pdf file as the system did not allow uploading a .cfg file. Just uploaded as a .txt file
[LDO_Nighthawk-SB_v1.0.txt](https://github.com/Frix-x/klippain/files/14289181/LDO_Nighthawk-SB_v1.0.txt)

mvdveer, 2024-02-15:
I have uploaded the two configs in zip file format in the correct directory structures. Hope this helps
[LDO_Nighthawk-SB.zip](https://github.com/Frix-x/klippain/files/14290183/LDO_Nighthawk-SB.zip)

mvdveer, 2024-02-15:
> you are always welcome to take a config as template and provide a configuration via a pull request to speed up the process :)

Here are the files in the correct directories in zip file format.
[LDO_Nighthawk-SB.zip](https://github.com/Frix-x/klippain/files/14290215/LDO_Nighthawk-SB.zip)

Benoitone, 2024-02-15:
> > you are always welcome to take a config as template and provide a configuration via a pull request to speed up the process :)
> 
> 
> 
> Here are the files in the correct directories in zip file format.
> 
> [LDO_Nighthawk-SB.zip](https://github.com/Frix-x/klippain/files/14290215/LDO_Nighthawk-SB.zip)
> 
> 

Why don't you do a PR?

mvdveer, 2024-02-15:
> Why don't you do a PR?

As I had not done a Pull request before, I had to do some reading - sorry for the inconvenience, but it has been done. Created a new pull request.

mvdveer, 2024-02-15:
> you are always welcome to take a config as template and provide a configuration via a pull request to speed up the process :)

Thank you - have done a PR - posted

Frix-x, 2024-02-26:
Everything is now done properly on #505 
Please let me know if it's ok for you now :)

closed 2024-02-27 by Frix-x