Frix-x/klippain #477: Add support to Fysetc Hotkeys

llambion72, 2024-02-15:
### Describe the feature or hardware support you'd like

Could you add Fystec hotkeys in the future? 
I've installed this addon on my voron 2.4 and it's very useful feature.

Surion79, 2024-02-16:
it would be helpful to give some links to what you like to be integrated :)

llambion72, 2024-02-18:
https://www.fysetc.com/products/fysetc-hot-key-board-voron-skirt-button-pcb-voron-skirt-klipper-pre-installed-pcb-board-with-neopixel-led-for-voron-v2-4-trident-switchwire-3d-printers

above link shows fysetc hotkey buttons product.

github link. There are klipper template config file and stls for buttons.
https://github.com/FYSETC/Hotkey/tree/main

Surion79, 2024-02-19:
looks doable. But I need to take a look. probably won't make it in the next release, which is released in a couple days apparently.

mjonuschat, 2026-08-30:
Implemented in PR #813

closed 2026-08-30 by mjonuschat