Frix-x/klippain #484: Unable to install due to package hash not matching

jasonmacdonald, 2024-02-19:
### Klippain branch

- [X] I confirm using the main branch

### Version

latest

### Describe the bug and expected behavior

I'm not a python guy, so I'm not sure if this is due to this repo or something else but, I'm unable to install due to a const error during install...

```
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    scipy==1.11.4 from https://www.piwheels.org/simple/scipy/scipy-1.11.4-cp39-cp39-linux_armv7l.whl#sha256=f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 4)):
        Expected sha256 f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739
             Got        aae0c31243227c148abc8beb83784934cd24a4292d01860b382ae00f6df09e4e
```
I've tried updating my PI, installing the package separately (which succeeded but I still get the error).

### Additional information and klippy.log

Using `wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash`

Full log
```
 $ wget -O - https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh | bash
--2024-02-19 12:27:15--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4730 (4.6K) [text/plain]
Saving to: ‘STDOUT’

-                                     100%[========================================================================>]   4.62K  --.-KB/s    in 0.003s

2024-02-19 12:27:15 (1.74 MB/s) - written to stdout [4730/4730]

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

python3-venv is already installed
libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

[SETUP] Virtual environment already exists. Continuing...
[SETUP] Installing/Updating K-Shake&Tune dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./klippain_shaketune-env/lib/python3.9/site-packages (24.0)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting GitPython==3.1.40 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/gitpython/GitPython-3.1.40-py3-none-any.whl (190 kB)
Collecting matplotlib==3.8.2 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 2))
  Using cached https://www.piwheels.org/simple/matplotlib/matplotlib-3.8.2-cp39-cp39-linux_armv7l.whl (10.7 MB)
Collecting numpy==1.26.2 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 3))
  Using cached https://www.piwheels.org/simple/numpy/numpy-1.26.2-cp39-cp39-linux_armv7l.whl (19.4 MB)
Collecting scipy==1.11.4 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 4))
  Downloading https://www.piwheels.org/simple/scipy/scipy-1.11.4-cp39-cp39-linux_armv7l.whl (66.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━╸━━━━━━━━━━━━ 46.0/66.0 MB 168.3 kB/s eta 0:01:59
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    scipy==1.11.4 from https://www.piwheels.org/simple/scipy/scipy-1.11.4-cp39-cp39-linux_armv7l.whl#sha256=f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739 (from -r /home/riloc/klippain_shaketune/requirements.txt (line 4)):
        Expected sha256 f719eaac73021b7ae4848de9ab76160b264c8a2726a5141090c178d0db131739
             Got        aae0c31243227c148abc8beb83784934cd24a4292d01860b382ae00f6df09e4e
```

jasonmacdonald, 2024-02-19:
And to make matters worse, the uninstall just deleted all of my configs files.

EricZimmerman, 2024-02-19:
Set up GitHub based backups!

jasonmacdonald, 2024-02-19:
> Set up GitHub based backups!

I have them.  That wasn't the point I was trying to make. What if I was just a novice user that didn't have back-ups? This isn't something the software should do when you click not to restore previous versions. It shouldn't delete what was there. Also, showing a warning of the fact the user has to reupload their config after the action has already been taken is also not a great end-user experience.

EricZimmerman, 2024-02-19:
I don't disagree.  Wonder how it would ever get deleted

EricZimmerman, 2024-02-19:
I wouldn't delete the backup folder even on an uninstall

jasonmacdonald, 2024-02-19:
I think it has something to do with it not fully installing. Just a guess though.

Surion79, 2024-02-19:
did you do a inplace dist upgrade without proper upgrading? did you use kiauh to install klipper? or did you used mainsailOS build?

Surion79, 2024-02-19:
looking at google search i found a lot of topics, like bad/corrupt download, intervening proxies or a bad cache which could cause such issues. Since Klippain does not require specific checksums (requirements.txt does not include those), it is an internal safety check of pip, which is causing the errors of the installation.

@EricZimmerman where is your guide for the github auto backup? :)

Surion79, 2024-02-19:
i consider the pip error not as a bug of klippain.
The uninstall needs apparently some additional checks.

jasonmacdonald, 2024-02-20:
> did you do a inplace dist upgrade without proper upgrading? did you use kiauh to install klipper? or did you used mainsailOS build?

This is a mainsailOS install with all updates installed, including `sudo apt-get update && apt-get upgrade -y`

Surion79, 2024-02-20:
Pip uses an own cache and if the download was manipulated by bad internet or proxy, those can happen. But it would be best if you google the error yourself. I don't have time to play google search for others. It is not a Klippain issue.

Frix-x, 2024-02-20:
Hello @jasonmacdonald,

I think I'm not sure about what happened and what you tried to do. Apparently, you tried to install the Shake&Tune module that is in another repository (https://github.com/Frix-x/klippain-shaketune). But you created an issue here that This one is the full Klippain configuration. The error you had is usually when there is some network problem. Doing these commands will probably fix it:
```bash
source ~/klippain_shaketune-env/bin/activate
python -m pip cache purge
deactivate
```

Regarding the uninstall process, I think it's because you called the full Klippain uninstall script while not using the full Klippain, so it apparently just cleaned up your config folder without finding the config. I'm sorry for that and will have to add some more checks, but as you may understand, I didn't think anyone would do that when I designed the script. Good thing you had some backups!

jasonmacdonald, 2024-02-20:
> Hello @jasonmacdonald,
> 
> I think I'm not sure about what happened and what you tried to do. Apparently, you tried to install the Shake&Tune module that is in another repository (https://github.com/Frix-x/klippain-shaketune). And you created an issue here but this repo is the full Klippain configuration. But nevermind, the error you have is usually when there is some network problem. Doing these commands will probably fix it and allow you to install Shake&Tune (if this is what you really want to do):
> 
> ```shell
> source ~/klippain_shaketune-env/bin/activate
> python -m pip cache purge
> deactivate
> ```
> 
> Regarding the uninstall process, I think it's because you called the full Klippain uninstall script while not using the full Klippain, so it apparently just cleaned up your config folder without finding the config. I'm sorry for that and will have to add some more checks, but as you may understand, I didn't think anyone would do that when I designed the script. Good thing you had some backups!

Thanks, Frix! Perhaps that is where I went wrong. I did have both repos open at the time and must not have noticed that I had gotten them mixed up. I totally understand you didn't account for that scenario. That's on me. Thanks for the guidance here.

closed 2024-02-20 by jasonmacdonald