Frix-x/klippain #492: Printer shuts down at the end of the print

camilloaddis, 2024-02-22:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-24-g4dd8a613

### Describe the bug and expected behavior

### BUG
Print goes fine, but at the end, when the default `END_PRINT` is called the printer does the `PARK` macro and then shuts down like if an `M112` command was issued. Then it automatically restarts and idles as if it was just turned on.

### Expected behaviour
Print ends and printer completes the `END_PRINT` macro without resetting, and then idles.

### Additional information and klippy.log

### Current configuration
Klippain
Klippain ShakeTune
Klipper TMC AutoTune
Auto Z Calibration
Crowsnest

Everything is updated

[image]

Attached the klippy.log created this way:
- flushed the old log > printer restarts
- printed a 20x20x2 mm rectangle
- print ends > printer halts
- manually call firmware_restart
- log downloaded

[klippy.log](https://github.com/Frix-x/klippain/files/14379703/klippy.log)

Frix-x, 2024-02-22:
Please look for your park position variable in the variables.cfg file. I'm pretty sure you still have it set to `-1, -1`. If this is the case, then change it to a proper position where you want to park the toolhead (usually over the purge bucket if you have one is a good idea) and that should solve your issue :)

camilloaddis, 2024-02-22:
Sorry, I think I need to add more infos:
the printer was working fine, everything (I think) was set up correctly: parking position, klicky probe position, some overrides, etc. In the last few days I made two changes:
- switched to klicky probe
- added the filter fan

Then I updated the components through mainsail and it had an issue with a cfg file missing. I reinstalled everything, and loaded the previous printer.cfg, mcu.cfg, variables.cfg, moonraker.conf and overrides.cfg.
Now everything is updated and fine but the problem is still there.

Also: if I home the printer and call the END_PRINT macro, it doesn't shut down. If I heat the bed and the nozzle to temperature and call the END_PRINT macro, again: works without issue. 
The issue shows only at the end of an actual print…

Benoitone, 2024-02-23:
Have a look at your slicer g-code?

camilloaddis, 2024-02-23:
If you mean the custom End G-Code set up in the slicer is just "END_PRINT".

The actual GCODE file sent to the printer ends with: 
```
G1 E-0.5 F1800
M204 S4000
G1 Z2.25 F27000
; stop printing object Shape-Box id:0 copy 0
M107
;TYPE:Custom
; custom gcode: end_filament_gcode
; Filament-specific end gcode 
;END gcode for filament
; custom gcode end: end_filament_gcode
; custom gcode: end_gcode
END_PRINT
; custom gcode end: end_gcode
M73 P100 R0
; filament used [mm] = 315.28
; filament used [cm3] = 0.76
; filament used [g] = 0.78
; filament cost = 0.02
; total filament used [g] = 0.78
; total filament cost = 0.02
; total layers count = 10
; estimated printing time (normal mode) = 3m 23s
```

And then few hundreds commented out lines with all superslicer config

camilloaddis, 2024-02-23:
[update]

I disconnected the wires of the filter from the MCU and the wires of the accelerometer from the OrangePi and it worked! Then I connected back again the filter wires and it still works, so maybe the accelerometer cables connected to the pi were messing up something.

camilloaddis, 2024-02-23:
Update:

**If I disable the filter from printer.cfg the issue is gone.**
Otherwise the filter remains 100% all the time during the print (material is correctly set to PLA and klippain is reading it right, and in the variables filter_speed is set to 0) and at the end the printer shuts down

Surion79, 2024-02-24:
Could you please add your printer.cfg? Or at least highlight which exact line you removed to get it working.

camilloaddis, 2024-02-24:
Here is the main config files (mcu, printer, overrides, variables, moonraker and crowsnest), so you can have the whole picture.

The line i commented out to make it working was under the `FILTER` section
```
# [include config/hardware/filters/exhaust_filter.cfg]
```

**BUT**
It worked for a few prints, then I tried to check what would have happened if I enabled it again and the issue was back again: filter il 100% during print if enabled, and even if disabled the filter the printer shuts down at the end of the print.

I've formatted the Orange Pi and reinstalled everything from scratch, reloaded the configuration but the issue is still there :(

[config-20240224-085541.zip](https://github.com/Frix-x/klippain/files/14392109/config-20240224-085541.zip)

Surion79, 2024-02-24:
parking position of 0,0 is correct for you? that would be front left on the bed, which makes printed part removal kind complicated

Surion79, 2024-02-24:
> Update:
> 
> **If I disable the filter from printer.cfg the issue is gone.** Otherwise the filter remains 100% all the time during the print (material is correctly set to PLA and klippain is reading it right, and in the variables filter_speed is set to 0) and at the end the printer shuts down

if filter is set to 0 for your used material, it should not start at all

Surion79, 2024-02-24:
"added the filter" does mean what? Nevermore? or "THE FILTER"? or the exhaust filter at the back?

camilloaddis, 2024-02-24:
> parking position of 0,0 is correct for you? that would be front left on the bed, which makes printed part removal kind complicated

Yeah is right - I plan to change this when I’ll add a purge bucket on the back but for now it works fine

camilloaddis, 2024-02-24:
> if filter is set to 0 for your used material, it should not start at all

Yep, but still it goes 100% a right after the bed starts heating till the end of the print, regardless of the material settings

camilloaddis, 2024-02-24:
> "added the filter" does mean what? Nevermore? or "THE FILTER"? or the exhaust filter at the back?

Exhaust filter at the back

Surion79, 2024-02-24:
ah, ok. that changes things. This works differently.

Surion79, 2024-02-24:
let me collect the stuff and write something about it

Surion79, 2024-02-24:
error found. The mapping in the MCU by default is wrong.

@Frix-x , @Benoitone I think we need to review all board user templates to verify EXHAUST_FAN is correctly linked. E.g. the Spyder 3 has linked a heater pin to it instead of a fan pin.

Surion79, 2024-02-24:
@camilloaddis can you please provide some information how your spyder is wired? so i can give you proper instructions to fix it

camilloaddis, 2024-02-24:
Sure!. The wiring mainly follows the Klippain template, with few differences:

```
### STEPPERS

STEPPER A               M1
STEPPER B               M0
STEPPER Z               M2
STEPPER Z1              M4
STEPPER Z2              M5
STEPPER EXTRUDER        M3

### HEATING ELEMENTS

HE0                     HEAT0       |   PB15
BED                     BED_OUT     |   PB4
TH0                     T0          |   PC0     
AUX                     T1          |   PC1
TB                      TB          |   PB0

### ENDSTOPS

X ENDSTOP               X-MAX       |   PA1
Y ENDSTOP               Y-MIN       |   PB13
Z ENDSTOP               Z-MIN       |   PA0

### PROBE

KLICKY                  Z-MAX       |   PA3

### FANS

PARTS FAN               FAN1        |   PA14
HOTENED FAN             FAN0        |   PA13
CONTROLLER FAN 1        FAN2        |   PB2
CONTROLLER FAN 2        FAN3        |   PB5
EXHAUST FAN             HEAT2       |   PB3

### LIGHTS

SB NEOPIXELS                        |   PD3
CHAMBER NEOPIXELS       Y-MAX       |   PA2

Surion79, 2024-02-24:
Can you please rewire the exhaust to FAN4? No board we have here uses heater connect for a fan. Was wondering about it when i saw it at fysetc wiring diagram.

camilloaddis, 2024-02-24:
I've rewired it to FAN4, and added this to `overrides.cfg`
```
[heater_fan exhaust_filter]
pin: MCU_FAN4
```
Same issues: printer shuts down after print and 100% exhaust fan all the time

Benoitone, 2024-02-24:
> error found (i think). The mapping in the MCU by default seems wrong.
> 
> 
> 
> @Frix-x , @Benoitone I think we need to review the spyder board user templates to verify EXHAUST_FAN is correctly linked. 

It's the default for the spider board on Voron config... https://github.com/VoronDesign/Voron-2/blob/Voron2.4/firmware/klipper_configurations/Spider/Voron2_Spider_Config.cfg

Surion79, 2024-02-24:
> It's the default for the spider board on Voron config... https://github.com/VoronDesign/Voron-2/blob/Voron2.4/firmware/klipper_configurations/Spider/Voron2_Spider_Config.cfg

Yes, but doesn't mean that it is best practice.

Surion79, 2024-02-24:
> I've rewired it to FAN4, and added this to `overrides.cfg`
> 
> ```
> 
> [heater_fan exhaust_filter]
> 
> pin: MCU_FAN4
> 
> ```
> 
> Same issues: printer shuts down after print and 100% exhaust fan all the time

It runs 100% all the time because the filter setting in the filament is used only for the nevermore. The exhaust filter you use is a completly different design within klipper with no macros.
If you want to have it per filament you need to flip the commented line from exhaust to nevermore and instead using Exhaust_fan you use filter_fan

camilloaddis, 2024-02-27:
update:

- I formatted the OrangePi and reinstalled everything from scratch (using the old cfg files) and this didn't solve the issue;
- I tried with a spare RaspberryPi 3, but again, same issue
- I copied the `END_MACRO` to `ovverrides.cfg` and stripped out all the "ifs", leaving only the calls to the actions I wanted to call, like this:

```
[gcode_macro END_PRINT]
gcode:
    PARK
    _TIP_SHAPING
    G92 E0
    G1 E-10 F2100
    TURN_OFF_HEATERS
    M107
    M400
    BED_MESH_CLEAR
    M84
    STATUS_LEDS COLOR="OFF"
```
and still it doesn't work. 
**BUT, if i comment out the last two lines the issue is gone!**

It's strange because if I remove only `M84` or `STATUS_LEDS COLOR="OFF"` the printer still don't crash, but if I call both the actions, the issue is back.

Can't understand why, but maybe this can help to understand if there's something wrong with the Klippain + Spider configuration, or anything else.

Surion79, 2024-02-27:
if you add a new script just with M84 and STATUS_LEDS would that crash it too? (before executing i suggest making it do something like already homed and lights on)

or what happens if you switch M84 and status_leds? maybe there is some issues with sagging, resulting in electric backlash by the motors

Surion79, 2024-02-27:
do you use can bus and if yes, how is it set up? Especially katapult make menuconfig would be very interesting. Some peeps had it complaining about shutting down and after some investigation i found out that someone posted a bogus can bus config, resulting in a boot issue if a pin was used, because it was also configured within katapult.

camilloaddis, 2024-02-27:
> if you add a new script just with M84 and STATUS_LEDS would that crash it too? (before executing i suggest making it do something like already homed and lights on)
>
> or what happens if you switch M84 [...]

I tried these scenarios:
```
gcode:
    PARK
    _TIP_SHAPING
    G92 E0
    G1 E-10 F2100
    TURN_OFF_HEATERS
    M107
    M400
    BED_MESH_CLEAR
    STATUS_LEDS COLOR="OFF"
    M84
```
Result: shutdown after print

```
gcode:
    M84
    STATUS_LEDS COLOR="OFF"
```
Result: shutdown after print

```
gcode:
    STATUS_LEDS COLOR="OFF"
    M84
```
Result: shutdown after print

> do you use can bus and if yes, how is it set up? Especially katapult make menuconfig would be very interesting. Some peeps had it complaining about shutting down and after some investigation i found [...]

No, just a Stealthburner PCB and a wire harness

Surion79, 2024-02-27:
i meant the two line macro to be called stand alone without a print. 
like:
Do a G32 (a.k.a. G28, QGL, G28 Z), set the leds for the print value and then execute the 2 line macro without print.
what is the result?
if it still shuts down, please add between the two lines a G4 S30 (this means, the gcode is "staled" for 30 seconds. to check if it is a race condition of the two commands

camilloaddis, 2024-02-27:
Ok, tried these: 

```
[gcode_macro DEBUG_SHUTDOWN]
gcode:
    G28
    Z_TILT_ADJUST
    G28 Z
    STATUS_LEDS COLOR=printing
    M84
    STATUS_LEDS COLOR="OFF"
```

Result: Shutdown

```
[gcode_macro DEBUG_SHUTDOWN]
gcode:
    G28
    Z_TILT_ADJUST
    G28 Z
    STATUS_LEDS COLOR=printing
    M84
    G4 S30
    STATUS_LEDS COLOR="OFF"
```
Result: Shutdown

```
[gcode_macro DEBUG_SHUTDOWN]
gcode:
    G28
    Z_TILT_ADJUST
    G28 Z
    STATUS_LEDS COLOR=printing
    G4 S30
    M84
    G4 S30
    STATUS_LEDS COLOR="OFF"
```
Result: Shutdown

But, to be honest, I think that the printer isn't waiting 30seconds with the `G4 S30` command. 
Indeed, if I run the first four lines of codes and then manually issue `M84` and then `STATUS_LEDS COLOR="OFF"` (typing the first command and pasting the second one as fast as I can) , there's no problem

Surion79, 2024-02-27:
Instead of S30 get P30000 for G4

camilloaddis, 2024-02-27:
> Instead of S30 get P30000 for G4

This time the printer correctly waited 30 seconds and then when calling the last command again it shutdown

Surion79, 2024-02-28:
so either M84 or STATUS_LEDS ? can you please try with the new 4.3.0 version of klippain?

camilloaddis, 2024-03-07:
I updated to klippain 4.3 and the problem now seems gone!

But, to be fair, here is the whole process:
- updated to 4.3 following the instructions on the release page, and the **printer wasn't working anymore**: the OrangePi wasn't reading the MCU most of the time, and sometimes the printer succesfully powered on but shut down after few seconds and couldn't do anything about it.
- I tried flashing again the Spider with 32kb bootloader and with no bootloader, without success: same issues (printer not woking)
- So I did a fresh install of everything klipper related: deleted everything from the ~ folder, reinstalled KIAUH and deleted all the services left. Reinstalled everything (Klipper, Moonraker, Mainsail, Crowsnest, KlipperScreen, Obico), then installed again Klippain and configured the printer from scratch (using some old values but not the old files). 
- I then run the same `DEBUG_SHUTDOWN` macro without issue! 
- So I reinstalled Timelapse and Autotune (enabled, not installed since is now included), and now it seems to work fine :)

Right now I did few tests and run few prints without issues, so I consider the problem gone – if anything happens again I'll update this thread.

Thanks for the support and for the amazing Klippain 🥖

Surion79, 2024-03-08:
glad that it worked out, please reopen the issue if this comes up again

closed 2024-03-08 by Surion79