Frix-x/klippain #499: Problem with CANCEL_PRINT call on gcode Error

elpopo-eng, 2024-02-24:
### Describe the feature or hardware support you'd like

If there is an error in the gcode file, `CANCEL_PRINT` is called.
However, `CANCEL_PRINT` also calls `SDCARD_RESET_FILE` , which cannot be called in a file. And so generates this error :
`!! SDCARD_RESET_FILE cannot be run from the sdcard`

Surion79, 2024-02-24:
What is the enhancement your request? Sounds more like a bug. 
What is your solution proposal?

elpopo-eng, 2024-02-24:
Just because I can't declare a bug because of form
In form we must declare that we are on main branch.

Surion79, 2024-02-24:
you can reflag it yourself :) you are a collaborator. Do you have a solution proposal for the bug?

Surion79, 2024-02-24:
just remove the `SDCARD_RESET_FILE` or make a copy of cancel print and use it for dedicated error on gcode called from the sdcard section of klipper?

elpopo-eng, 2024-02-26:
@Benoitone Why did you include CANCEL_PRINT in gcode_error?
I'm not a big fan of disabling the heater when you have a gcode error because you can't restart quickly after the error is fixed.

Surion79, 2024-02-26:
> @Benoitone Why did you include CANCEL_PRINT in gcode_error?
> 
> I'm not a big fan of disabling the heater when you have a gcode error because you can't restart quickly after the error is fixed.

I guess that is something for an easy variable to fix without altering the functionality 😃

Frix-x, 2024-02-28:
This should be solved on the last commit from the main branch

closed 2024-02-28 by Frix-x