Frix-x/klippain #502: X axis shaper graphs not created -

zwergigor, 2024-02-25:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.2.0-24-g4dd8a613

### Describe the bug and expected behavior

Hi, i get this failure all time. Tryed in my working klippel installation.
And now also in my new installation.
Setup Klipper, Manisail, Mooknraker
Install Klippain and try  AXES_SHAPER_CALIBRATION

Result:
`03:00 Probing point (150.000, 150.000, 20.000)
03:00 Command {plot_graph} finished
03:00 ~~~~~~~~~~~~~^^~
TypeError: unsupported operand type(s) for ** or pow(): 'function' and 'int'
03:00 df = math.sqrt(1. - damping_ratio**2)
03:00 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/pi/klipper/klippy/extras/shaper_defs.py", line 18, in get_zv_shaper
03:00 shaper = shaper_cfg.init_func(test_freq, damping_ratio)
03:00 return method(*args)
^^^^^^^^^^^^^
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 260, in fit_shaper
03:00 File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 74, in background_process_exec
03:00 shaper = self.background_process_exec(self.fit_shaper, (
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
03:00 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/pi/klipper/klippy/extras/shaper_calibrate.py", line 330, in find_best_shaper
03:00 shaper, all_shapers = helper.find_best_shaper(calibration_data, max_smoothing, print)
03:00 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/pi/klippain_config/scripts/is_workflow/graph_shaper.py", line 82, in calibrate_shaper_with_damping
03:00 performance_shaper, shapers, calibration_data, fr, zeta = calibrate_shaper_with_damping(datas, max_smoothing)
03:00 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/home/pi/klippain_config/scripts/is_workflow/graph_shaper.py", line 335, in shaper_calibration
03:00 fig = shaper_calibration([new_file], KLIPPER_FOLDER)
03:00 ^^^^^^^^^^^^^^^^^^
File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 116, in get_shaper_graph
03:00 fig, png_filename = get_shaper_graph()
03:00 File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 217, in main
03:00 main()
03:00 Traceback (most recent call last):
File "/home/pi/printer_data/config/scripts/is_workflow/is_workflow.py", line 231, in
03:00 Running Command {plot_graph}...:
03:00 X axis shaper graphs generation...`

### Additional information and klippy.log

Running Klipper on
Raspberry Pi 4 Model B Rev 1.2 (4GB)
OS: Linux - 6.1.0-rpi8-rpi-v8 (bookworm) - but occurs also in bullseye

zwergigor, 2024-02-25:
[Uploading klippy (18).log…]()

Frix-x, 2024-02-25:
Hello, this is introduced by a Klipper change from the 17Feb. S&T was updated but since Klippain incorporate a custom version of S&T it doesn't have the fix for now. I do plan to release a new Klippain version with a proper integration of S&T (it's already done in the develop branch)  very soon. Probably tonight if I manage to merge the last PRs

M3HNGRY, 2024-02-26:
Same issue for me as well! Just waiting for the updates....

closed 2024-02-26 by Frix-x