Frix-x/klippain #509: Upgrade to the latest Klippain Bug?

M3HNGRY, 2024-02-28:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.0-0-geeb49160

### Describe the bug and expected behavior

Updated to the latest Klippain. Octopus, Tap, SB2240, BTT USB2CAN, and Pi4.  Able to input shape x and y now. When I go to print. 

1. Unknown gcode_macro_variable "adaptive_primeline" 
2. No extruder retraction because extruder temp is (48.88) is lower then min_extrude_temp (172)
3. SDCARD_RESET_FILE cannot be run from the sdcard

[image]

My Orca slicer start gcode. 

STARTPRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] MATERIAL=[filament_type] CHAMBER=[chamber_temperature] SIZE={first_layer_print_min[0]}{firstlayer_print_min[1]}{firstlayer_print_max[0]}{first_layer_print_max[1]} 

### Additional information and klippy.log

[moonraker (1).log](https://github.com/Frix-x/klippain/files/14427814/moonraker.1.log)
[klippy (12).log](https://github.com/Frix-x/klippain/files/14427815/klippy.12.log)

joseaneuro, 2024-02-28:
I'm also having the same issue. I looked in the prime_line.cfg file and can't find a reference to ADAPTIVE_PRIMELINE. I'm wondering if the PRIMELINE macro was supposed to be renamed to ADAPTIVE_PRIMELINE.

M3HNGRY, 2024-02-28:
The other klippy log was empty. Here is current one. 

[klippy (14).log](https://github.com/Frix-x/klippain/files/14429054/klippy.14.log)

joseaneuro, 2024-02-28:
@Frix-x Found the fix!

The START_PRINT macro was missing "variable_adaptive_primeline: 1" inside it. Redifining it in the overrides to add that fixes it.

[image]

Frix-x, 2024-02-28:
Adaptive prime line variable was indeed added a couple of hours ago. I still need to fix the rest. Thanks for raising this issue :)

Frix-x, 2024-02-28:
Everything should be fixed except the Neopixel part. Do you have more info about the problem?

M3HNGRY, 2024-02-28:
Frix, 

I have updated and now printing. I think we are good to go now. I will keep an eye on the neopixel part and open a new ticket if there is any more errors. . Thanks!

Frix-x, 2024-02-28:
Thanks! I'll close this issue then :)

closed 2024-02-28 by Frix-x