Frix-x/klippain #526: Failed to detect git branch & shake&tune fails to install

Svennixx, 2024-03-03:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.0

### Describe the bug and expected behavior

I have installed a fresh klipper on my new raspberry pi 4, installed klippain without issues really. Everything works except for the moonraker updates and shake&tune (even though it is installed according to the install.sh).
[image]
[image]
[image]

Frix-x, 2024-03-04:
Can you share your printer.cfg file? Because usually when this happens it's because you did not used any of the accelerometer files as they are the one that make S&T macro automatically available

Svennixx, 2024-03-04:
> Can you share your printer.cfg file? Because usually when this happens it's because you did not used any of the accelerometer files as they are the one that make S&T macro automatically available

I feel so dumb, thats it! 
I didn't include that because I'm using the new revision of the beacon probe which has an accelerometer build in. And klippain doesn't have file for that yet. 

Shake&Tune is fixed but klippain git is still not being found for update manager.

Frix-x, 2024-03-04:
Ok so first, this is a good thing :) Then, can you share your beacon accelerometer definition to include it in Klippain ?

Regaridng the "failed to detect repo url", etc... I'm not sure about what could cause this. Can you use SSH and do in the klippain_config folder "git status" and report the result here?

Svennixx, 2024-03-04:
yeah sure this is all you need to use the beacon (revH) its build in accelerometer:
`[resonance_tester]
accel_chip: beacon`
No pin definitions or adxl345 setup needed. 

git status results:
[image]

Svennixx, 2024-03-04:
Okay not sure what happened, I clicked on the refresh button at the update manager on the UI and it just fixed itself.
Hopefully it will stay like this even on reboots.

closed 2024-03-04 by Svennixx