Frix-x/klippain #537: STATUS_LEDS not called when using neopixel_caselight_effects.cfg without status_leds*.cfg

qratz, 2024-03-07:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3

### Describe the bug and expected behavior

when including neopixel_caselight_effects.cfg without any of the status_leds config files, caselight effects are not triggered in any of the macros, like START_PRINT, END_PRINT, etc.
The call to STATUS_LEDS in those scripts is only processed if variable_status_leds_enabled is set to True.
As a quick fix i replaced
`{% set status_leds_enabled = printer["gcode_macro _USER_VARIABLES"].status_leds_enabled %}`
with 
`{% set status_leds_enabled = printer["gcode_macro _USER_VARIABLES"].status_leds_enabled or printer["gcode_macro _USER_VARIABLES"].status_leds_caselight_enabled %}`
in all those scripts.

Frix-x, 2024-03-07:
This is not a bug but "normal":
- The goal of status leds is for some devices to show the printer status with colors. Usually on a Voron printer, it's some leds on the StealthBurner, but this can be anything else
- The caselights leds are not using colors and so on to avoid a "Disco light show" and just illuminate the machine as standard lights should do

However, you are right, the caselight effects I don't know what they should do... But your fix would break the two previous point because standard caselights would also get the colors...

EricZimmerman, 2024-03-07:
Id think most peoples case lights are all white vs RGB?

Benoitone, 2024-03-07:
> This is not a bug but "normal":
>
>     * The goal of status leds is for some devices to show the printer status with colors. Usually on a Voron printer, it's some leds on the StealthBurner, but this [...]

For me `status_leds_enabled` AND `status_leds_caselight_enabled` use RGB instead of `light_enabled` witch use white leds.
So I think @qratz is right and is fix can be a good one???
I think `status_leds_caselight_enabled` has been introduce to be able to control neopixel caselight and/or status_led in `STATUS_LEDS` macro

qratz, 2024-03-08:
I was looking at this issue when switsching my V0.2 to klippain. It has the "Disco on a Matchstick"-PCB boards and a BARF-style board for the kirigami bed mount. To get the lights actually do something except lighting up, i applied that quick adjustment. I actually thought that the idea of caselight_effects was to actually use the effects, especially because of this section in status_leds.cfg:
```
    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_caselight_enabled%}
        {% if printer["gcode_macro _USER_VARIABLES"].status_leds_effects_enabled == False %}
            _SET_ALLLEDS_BY_NAME LEDS="caselight" COLOR={status_color[color].caselight} TRANSMIT=1
        {% else %}
            SET_LED_EFFECT EFFECT={"cl_" + status_color[color].caselight} REPLACE=1 FADETIME=0.5
        {% endif %}
    {% endif %}
```
So there is a call to set "cl_" effect, but it is basically only used startup.cfg to set the caselight effect to READY. That is why i thought there i was something missing there.

Benoitone, 2024-03-08:
All the case light effect are based on this, like QGL, heating, pause, printing etc...

Benoitone, 2024-03-08:
Maybe must add
variable_status_leds_enabled: True
In the case lights hardware config files?

qratz, 2024-03-08:
> Maybe must add variable_status_leds_enabled: True In the case lights hardware config files?

If you are not using actual status_leds, but only caselights, this leads to SET_LED_EFFECT calls to statuslights which are not defined.

Benoitone, 2024-03-08:
so maybe add a new user_variable like `variable_neopixel_leds_enabled = True` in all lights using neopixel leds:
`neopixel_caselight.cfg`
`neopixel_caselight_effects.cfg`
`status_leds.cfg`
`status_leds_effects.cfg`
`status_leds_rainbow_barf.cfg`
`status_leds_rainbow_barf_effects.cfg`

and use `neopixel_leds_enabled` instead of `status_leds_enabled` in all macros?

Benoitone, 2024-03-09:
#541

Surion79, 2024-03-11:
i have neo pixel sticks in my printer and only use them for white and off. how can this be achieved with this change? I don't need a fully colored printer for the whole print. i like it white for better timelapse videos

EricZimmerman, 2024-03-11:
Don't use the LED effects cfg?

Surion79, 2024-03-11:
i want to use it for toolhead but not for case light?

Benoitone, 2024-03-12:
This PR is only for the issue happening when only one of the neopixel caselight is include in printer.cfg without any status_led

Benoitone, 2024-03-12:
But if you only want white color for caselight neopixel you can include neopixel_caselight.cfg in your printer.cfg and override _LEDS_COLORS_DEFINITION macro

teookie, 2024-03-16:
I noticed this problem on a new printer with RGB caselights and no stealthburner LED's. Uncommenting neopixel_caselight in printer.cfg enables all the caselight effects, but they are never called by PRINT_START or other macro's because the macros all have IF statements that only trigger if "status_leds_enables" is true, but not "status_leds_caselights".

Seems like making the change below would be minimal impact but would fix the problem:

ADD TO MACROS:
`{% set status_leds_caselights_enabled = printer["gcode_macro _USER_VARIABLES"].status_leds_caselight_enabled %}`

ADD AN "OR" CONDITION TO STATUS LEDS CALLS IN MACROS:
```
{% if status_leds_enabled or status_leds_caselight_enabled %}
        STATUS_LEDS COLOR="BUSY"
    {% endif %}
```

(I should say I have no idea if the syntax for the "or" statement I'm proposing is correct or not, but I think you get the idea. :) )

closed 2024-06-16 by github-actions[bot]

Didier-T, 2024-11-20:
Bonjour,
je me permet de réouvrir le sujet car il me semble que mon souci y est lié.

Dans le cadre de ma configuration j'aimerai pouvoir utiliser le caselight sans led effect et le status led avec led effect, mais actuellement la variable d'activation du led effect est commune aux deux type d'éclairage.

La solution la plus simple serait de modifier deux fichier :

1.  "config/hardware/lights/neopixel_caselight_effects.cfg" ligne n°6
    -  de variable_status_leds_effects_enabled=True
    -  a variable_status_leds_effects_caselight_enabled=True

2.  "macros/hardware_functions/status_leds.cfg" ligne n°215
    - de {% if printer["gcode_macro _USER_VARIABLES"].status_leds_effects_enabled == False %}
    - a  {% if printer["gcode_macro _USER_VARIABLES"].status_leds_effects_caselight_enabled== False %}

Par avance merci de prendre un peu de temps pour cette légère amélioration