Frix-x/klippain #542: Support multiple klipper instances

jumika, 2024-03-09:
### Describe the feature or hardware support you'd like

The install script currently have the klipper service and the paths hard coded relative to the home directory. I think it would be easy to implement choosing between instances by making the klipper path, config path and klipper service name parameters of the install / uninstall scripts or using a dotenv approach.
I already had success installing klippain on a multiple instance klipper host by modifying the install script, but it would be nice to support this out of the box.

Surion79, 2024-03-10:
if you have a working script, you can open up a PR :)

Frix-x, 2024-03-11:
There's already some work going on this subject with a PR that is almost ready here: https://github.com/Frix-x/klippain/pull/455

jumika, 2024-03-12:
I've tried this again on the weekend with another printer and found out there are a few more places to change the paths. I'm sure #455 is already more refined than my solution so i'm closing this.

closed 2024-03-12 by jumika