Frix-x/klippain #548: Change [output_pin] to [pwm_cycle_time] for PWM controlled outputs

norris108, 2024-03-11:
### Describe the feature or hardware support you'd like

As per the [Klipper Configuration Changes](https://www.klipper3d.org/Config_Changes.html#changes), 20240123, the `output_pin SET_PIN CYCLE_TIME` parameter has been removed in favour of the new [`pwm_cycle_time`](https://www.klipper3d.org/Config_Reference.html#pwm_cycle_time) module

### Additional context or information

I'm trying to get the beeper on my BigTreeTech mini 12864, but as the beeper is currently defined in `config/hardware/displays/BTT_mini12864.cfg` as:

```
[output_pin beeper]
pin: EXP1_1
```

I'm unable to override this in `overrides.cfg` with something like this:

```
[pwm_cycle_time beeper]
pin: EXP1_1
value: 0
shutdown_value: 0
cycle_time: 0.0005
scale: 1
```

as Klipper errors on restart with `pin EXP1_1 used multiple times in config`

Surion79, 2024-03-11:
can you create a PR?

Surion79, 2024-03-11:
well, have you tested your recommendation?

norris108, 2024-03-11:
I have on a non-Kllippain installation

I can't test it on my Klippain installation as Klipper errors on restart with `pin EXP1_1 used multiple times in config`

Surion79, 2024-03-11:
it is enough if you test it on non-klippain

Surion79, 2024-03-11:
i guess it requires rewriting all beepers in all mini12864 displays?

norris108, 2024-03-11:
According to the [Klipper Configuration Changes](https://www.klipper3d.org/Config_Changes.html#changes), any `[output_pin]` that is PWM controllable should be changed to use the new [`[pwm_cycle_time]`](https://www.klipper3d.org/Config_Reference.html#pwm_cycle_time) module

Surion79, 2024-03-11:
could you please check the linked PR i created? thanks

norris108, 2024-03-11:
Wow! That was quick work!

PR looks good to me

Surion79, 2024-03-11:
replacing stuff isn't hard. Since i don't use the display i can't test it