Frix-x/klippain #551: Move out of range during homing.

maxduracell, 2024-03-13:
### Klippain branch

- [X] I confirm using the main branch

### Version

4.3.1

### Describe the bug and expected behavior

Setting up a new v0.2.  Utilizing the 120mm preset, no probe, and sensorless homing. 
Homing moves X, then Y, then faults for a “move out of range” error.  Console reports the head is being moved to clear the area for a probe. The homing macro is requesting the printer to move an axis to -1, but the axis mins for the 120mm preset are 0. 
Expected behavior:
The macro to not call moves out of range and to not reference a probe that doesn’t exist. 

This is my first v0 so I’m not sure if the printer is even physically capable of moving into the negative. I don’t have the bed on yet to check as I’m waiting on some parts before I install it. Im more than happy to answer any follow up questions. Thanks!

maxduracell, 2024-03-13:
Messed with this some more and got it working, but I still think that it shouldn't be reacting to a non-existent dock.  The fix was to set variable_zendstop_position from -1, -1.

maxduracell, 2024-03-13:
Where does it say to do that? Maybe I missed something during the setup. I haven’t had a printer that didn’t use a probe for the Z end stop in a long time, but wouldn’t the assumption be that if you aren’t using a probe that the end stop would be on the axis itself and not require a position to be set?
In this specific instance at least, you can set the end stop position to anywhere in your range of motion. It doesn’t matter. I set mine to 60,120 because I find it aesthetically pleasing. 
if you go through the homing override macro, it does check to see if there’s a probe, but then no matter what, it calls a macro that moves the head to avoid a dock.  Just seems odd to me, but this is my first time really diving into klippain as I wanted to try out the tuning macros.

EricZimmerman, 2024-03-14:
That comment I put is right from variables.cfg

You don't have to make it centered for sure.

closed 2024-03-14 by maxduracell