Frix-x/klippain #560: When I am uninstall klippain all the files in config folder are gone!

adct2luv, 2024-03-19:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-9-g1c5c6623

### Describe the bug and expected behavior

I dont know why this happen
[image]

EricZimmerman, 2024-03-19:
What's in the config folder? What's in the klippain backup folder? 

Did you make your own backup before installing klippain?

adct2luv, 2024-04-19:
> 
> Did you make your own backup before installing klippain?

Yes, I am so lucky that I have the backup. but I have to setting up new from the ground

closed 2024-06-05 by github-actions[bot]