Frix-x/klippain #573: Add autotap for Voron tap machines

smwoodward, 2024-03-30:
### Describe the feature or hardware support you'd like

I've been using https://github.com/Anonoei/klipper_auto_tap for a while on my setup.  It's listed as being deprecated, but it might be something to still integrate into the Voron Tap setups.

EricZimmerman, 2024-03-30:
If it's depreciated it doesn't sound like a good idea to add. 

I never saw the point of that software either. 

You can always just add it manually in overrides

smwoodward, 2024-03-30:
> If it's depreciated it doesn't sound like a good idea to add.
> 
It's deprecated because the original writer moved away from Tap, so he's not going to do any more development on it.  
> I never saw the point of that software either.
> 
It's the same point as auto zeroing any of the other Z probes
> You can always just add it manually in overrides
I already said I've been using it for a while in my setup, so i've already manually added it in overrides, but isn't the point of Klippain to help make klipper setup easier?

Surion79, 2024-04-10:
I think this is really something for individual overrides.

closed 2024-05-26 by github-actions[bot]