Frix-x/klippain #574: PARK does not take Z_OFFSET into account when calculating max_z

GreenPlasticWaterCan, 2024-03-30:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-10-g1d9a2b53

### Describe the bug and expected behavior

PARK right now does not take into account a positive Z offset, potentially resulting in a move out of range. https://github.com/Frix-x/klippain/blob/1d9a2b5344e230818129f1481b1fdffc61af1257/macros/base/park.cfg#L18

`Move out of range: 60.000 60.000 126.000 [0.000]`

Reproducer on a V0 with 120x120x120 printbed and lift_z at > 20:

`G28
G90
SET_GCODE_OFFSET Z=5 MOVE=1
G1 X60 Y60 Z100
PARK`

Changed made to macro to fix:

` {% set max_z = printer.toolhead.axis_maximum.z - printer.gcode_move.homing_origin.z |float %}`

Thanks @reemo3dp for helping me debug and providing the fix.

Surion79, 2024-05-31:
i added a PR with your fix. thanks :)

Karolas123, 2024-06-03:
Hello, i still have thiss issue, how i can repair this?

Frix-x, 2024-06-18:
This was merged today :)