Frix-x/klippain #575: Shake&Tune fails to install mathplotlib

mmemetea, 2024-04-05:
### Klippain branch

- [X] I confirm using the main branch

### Version

N.A. as I can't finish a proper installation

### Describe the bug and expected behavior

The installation script doesn't complete and throws the error message "ERROR: No matching distribution found for matplotlib==3.8.2" before stopping.

### Additional information and klippy.log

wget -O - https://raw.githubusercontent.com/Frix-x/klippain/main/install.sh | bash
--2024-04-05 12:23:33--  https://raw.githubusercontent.com/Frix-x/klippain/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12163 (12K) [text/plain]
Saving to: ‘STDOUT’

-                                             100%[===============================================================================================>]  11.88K  --.-KB/s    in 0.002s

2024-04-05 12:23:33 (6.50 MB/s) - written to stdout [12163/12163]

======================================
- Klippain install and update script -
======================================

[PRE-CHECK] Klipper service found! Continuing...

[DOWNLOAD] Klippain repository already found locally. Continuing...

[BACKUP] Backup of current user config files done in: /home/pi/klippain_config_backups/2024_04_05-122333

[INSTALL] Installation of the last Klippain config files
[POST-INSTALL] Restarting Klipper...
--2024-04-05 12:23:34--  https://raw.githubusercontent.com/Frix-x/klippain-shaketune/main/install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4730 (4.6K) [text/plain]
Saving to: 'STDOUT'

-                                             100%[===============================================================================================>]   4.62K  --.-KB/s    in 0.01s

2024-04-05 12:23:34 (472 KB/s) - written to stdout [4730/4730]

=============================================
- Klippain Shake&Tune module install script -
=============================================

[PRE-CHECK] Klipper service found! Continuing...

python3-venv is already installed
libopenblas-dev is already installed
libatlas-base-dev is already installed
[DOWNLOAD] Klippain Shake&Tune module repository already found locally. Continuing...

[SETUP] Virtual environment already exists. Continuing...
[SETUP] Installing/Updating K-Shake&Tune dependencies...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in ./klippain_shaketune-env/lib/python3.7/site-packages (24.0)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting GitPython==3.1.40 (from -r /home/pi/klippain_shaketune/requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/gitpython/GitPython-3.1.40-py3-none-any.whl (190 kB)
ERROR: Ignored the following versions that require a different python version: 3.6.0 Requires-Python >=3.8; 3.6.0rc1 Requires-Python >=3.8; 3.6.0rc2 Requires-Python >=3.8; 3.6.1 Requires-Python >=3.8; 3.6.2 Requires-Python >=3.8; 3.6.3 Requires-Python >=3.8; 3.7.0 Requires-Python >=3.8; 3.7.0rc1 Requires-Python >=3.8; 3.7.1 Requires-Python >=3.8; 3.7.2 Requires-Python >=3.8; 3.7.3 Requires-Python >=3.8; 3.7.4 Requires-Python >=3.8; 3.7.5 Requires-Python >=3.8; 3.8.0 Requires-Python >=3.9; 3.8.0rc1 Requires-Python >=3.9; 3.8.1 Requires-Python >=3.9; 3.8.2 Requires-Python >=3.9; 3.8.3 Requires-Python >=3.9; 3.8.4 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement matplotlib==3.8.2 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2, 3.5.3)
ERROR: No matching distribution found for matplotlib==3.8.2

EricZimmerman, 2024-04-05:
Upgrade your operating system. I have a writeup on my GitHub profile under vorontools

Frix-x, 2024-04-17:
Yes this is a known issue that actually arise from Shake&Tune requirements: it needs Python >=3.9 that is not available in Debian Buster but is in Bulleyes and Bookworm. So just upgrade your OS and it will be solved

closed 2024-04-17 by Frix-x