Frix-x/klippain #576: Wrong pin on MCU definition of Bigtreetech SKR2

Andi3D-tech, 2024-04-06:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-10-g1d9a2b53

### Describe the bug and expected behavior

I wanted to connect the Bigtreetech Mini12864 to my BTT SKR2 via EXP1 and EXP2.

Unfortunately the display remained black.

Neopixel and reset button worked. Also the beeper.

_After a long search I found the error:_

Under config/mcu_definitions/main there is the **BTT_SKR_2.cfg**

```
# EXP1 header
 EXP1_1=PC5 , EXP1_2=PB0 ,
 EXP1_3=PB1 , EXP1_4=PB9 ,
 EXP1_5=PE10 , EXP1_6=PE11 , # Key in the socket on this side
 EXP1_7=PE12 , EXP1_8=PE13 ,
 EXP1_9=<GND> , EXP1_10=<5V> ,
```

EXP1_4=PB9 should be changed to **EXP1_4=PE9.**

The pin PB9 points to the I2C header.

Thanks!!!

tsk-2222, 2024-04-09:
Yup. It's a typo. Thanks for the eyeballs.

I'm a git/github newb and I didn't manage my branches properly so now I'm lost at trying to create a proper push for this simple fix.
Can someone help:  
https://github.com/tsk-2222/klippain/commit/ca6d9e6a744019c83607710811783c66000d3c07

Surion79, 2024-06-21:
completed by merged PR

closed 2024-06-21 by Surion79