Frix-x/klippain #577: The install didn't work with old mainsailos

bofh69, 2024-04-06:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-10-g1d9a2b5

### Describe the bug and expected behavior

I ran the install script, it looked like everything was done correctly, but klipper still ran with my old config.

I have a rpi & klipper installation based on mainsailos from two years ago. It is starting klipper via systemd and a `/etc/systemd/system/klippy` file with these lines:
```
[Service]
...
Environment=KLIPPER_CONFIG=/home/pi/klipper_config/printer.cfg
ExecStart= /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py ${KLIPPER_CONFIG} -l ${KLIPPER_LOG}  -a ${KLIPPER_SOCKET}
```

so obviously it didn't find the new config in `$HOME/printer_data/config/printer.cfg`.

I had a $HOME/printer_data/config, it was a symlink to `$HOME/klipper_config`.

Its probably not worth updating the install script to handle this, newer mainsailos seem to use `/home/pi/printer_data/printer.cfg`, but  maybe add something in the documentation about this? Just running `ps x | grep '[/]klippy/klippy.py'` showed me the path it was using.

EricZimmerman, 2024-04-06:
i would suggest reinstalling here. it will fix far more things for you than updating klippain to work with outdated directories. use this:

https://github.com/EricZimmerman/VoronTools/blob/main/OSUpgrade.md

especially since you are just starting with klippain

bofh69, 2024-04-06:
@EricZimmerman I've already updated the os (through apt) and most other things, the old klippy service file seems to have been kept though.

EricZimmerman, 2024-04-06:
thats the wrong way to do it.

start from scratch, or uninstall klipper via kiauh then reinstall to get the right paths.

Frix-x, 2024-04-17:
Yes this is a known issue but the thing is that it changed more than two years ago from now and there is almost no one still runnning the old paths. You can just use KIAUH to uninstall and resintall Klipper + Moonraker. This will fix your paths.

But as Eric said, there is actually more and you will probably hit another issue next that arise from Shake&Tune requirements: it needs Python >=3.9 that is not available in Debian Buster but is in Bulleyes and Bookworm.

So just upgrade your OS to a newer version and you will have everything solved at once :)

closed 2024-06-01 by github-actions[bot]