Frix-x/klippain #583: customizable END_PRINT, similar to START_PRINT

teookie, 2024-04-24:
### Describe the feature or hardware support you'd like

Would be great if custom modules could be called in the end_print macro.

Surion79, 2024-05-09:
how many would you like and where should they be placed? It is a bit more than just adding some custom calls. it would require some sort of endprint_actions like for start print

Surion79, 2024-06-21:
@Frix-x is there a need for this? it seems a bit like a bigger implementation

rajenki, 2024-07-16:
@Surion79 I'd love this as well. 3 custom actions for end and cancel print seems great, similar to start print (including endprint_actions/cancelprint_actions).

Surion79, 2024-07-16:
before implementing something like that I require more input:
which macro commands in endprint should be put in which submodules for endprint? 
it needs to be divided sufficiently so stuff can be excluded by being removed from endprint_actions.

rajenki, 2024-07-16:
I'm relatively new to Klippain still, so I'm probably not the best person to advise, but if you're asking how to break up the end/cancel print macros, I'd say the existing blocks (retract, turn off heaters, tach check, mesh clear, disable motors, filter, lights, reset filament sensor) would be a good start?

jonathandann, 2024-07-30:
I'd like to be able to customise a (safe) x_y_z travel (including speed) so the toolhead can move away quickly from the park to avoid stringing. Similar to the move defined in https://github.com/VoronDesign/Voron-2/blob/Voron2.4/firmware/klipper_configurations/SKR_1.4/Voron2_SKR_14_Config.cfg#L574

This might workable enough as some variables, instead of a full-blown action as in print_start.

LuisWinter, 2024-09-01:
I would like something like this as well, or alternatively, have skew profiles in the base macro.