Frix-x/klippain #585: Add the Pause at Layer Functionality for Mainsail

Elias23, 2024-04-29:
### Describe the feature or hardware support you'd like

I'd love to see pause at layer functionality from mainsail to be integrated into klippain.
https://github.com/mainsail-crew/mainsail-config#new-feature-pause-at-next-layer-and-pause-at-layer-

Elias23, 2024-05-10:
So, as I understand it, there is a new option in mainsail to schedule pauses while printing for either the next layer change or at a certain layer. It requires, as far as I know, the slicer config you mentioned plus 2 new macros(`SET_PAUSE_NEXT_LAYER` and `SET_PAUSE_AT_LAYER`) and a redefinition of the `SET_PRINT_STATS_INFO` macro.
New GUI option:
[image]
Feature in action via overrides:
[image]
I assume it would be pretty straight forward to add those macros to klippain, but I might be missing something there.

Surion79, 2024-06-20:
i created a PR to check. i added the functions as described in the linked repo

Elias23, 2024-06-24:
Thank you very much.

closed 2024-06-24 by Elias23