Frix-x/klippain #586: Parking after a tall print causes Move out of range error

shaxs, 2024-05-02:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-10-g1d9a2b53

### Describe the bug and expected behavior

When I was printing a tall cylinder shape that maxed out at 220mm, after finishing the print it tried to park but received a  Move out of range error. I have a 300mm 2.4. We diagnosed in in Discord here: https://discord.com/channels/460117602945990666/1233833838183190589

Looks like the code doesn’t account for offset due to bed_mesh/z_offset.

GreenPlasticWaterCan, 2024-05-06:
I raised an issue for this exact same issue.
Added a fix in the message too.

https://github.com/Frix-x/klippain/issues/574

Surion79, 2024-05-31:
added a PR with a fix as described by @GreenPlasticWaterCan

Frix-x, 2024-06-18:
This was merged today :)

closed 2024-06-18 by Frix-x

shaxs, 2024-06-18:
Awesome thank you!