Frix-x/klippain #590: END_PRINT macro static filter fan

bananensoftware, 2024-05-23:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-11-g4ff6f5b4

### Describe the bug and expected behavior

The filter macro sets the filter_name for using the correct fan dynamically:
```
[gcode_macro START_FILTER]
gcode:
    {% set SPEED = params.SPEED|default(1)|float %}

    {% set filter_name = printer["gcode_macro _USER_VARIABLES"].filter_name %}
    SET_FAN_SPEED FAN={filter_name} SPEED={SPEED}
```

But not in the END_PRINT macro, which breaks if a diffrent name then filter is used:
```
    {% if filter_enabled %}
        {% if printer['fan_generic filter'].speed > 0 %}
            {% set FILTER_TIME = params.FILTER_TIME|default(filter_default_time)|int %}
            START_FILTER SPEED=1
            UPDATE_DELAYED_GCODE ID=_STOP_FILTER_DELAYED DURATION={FILTER_TIME}
        {% endif %}
    {% endif %}
```

Surion79, 2024-05-31:
you mean the if-line?

bananensoftware, 2024-06-09:
yes, the current implementation makes it pointless to have configurable filter name.

rajenki, 2024-07-16:
Same issue as mentioned in #610

closed 2024-08-30 by github-actions[bot]