Frix-x/klippain #593: Make end print retract amount a variable

wight554, 2024-06-01:
### Describe the feature or hardware support you'd like

Currently primeline purges 18mm and end_print retracts 10mm which creates a blob at the start of primeline. In my setup my fan duct keeps grabbing that blob when purgeline is in X direction which melts it after a while

Similar to KAMP's https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/blob/main/Configuration/KAMP_Settings.cfg#L26 we can benefit even more from having variable accessible both from prime line and end_print

Lines of code to change:
https://github.com/Frix-x/klippain/blob/main/macros/helpers/prime_line.cfg#L113
https://github.com/Frix-x/klippain/blob/main/macros/base/end_print.cfg#L28

Surion79, 2024-06-20:
so these are practically two changes? One for end print and one for prime line?

wight554, 2024-06-20:
@Surion79 yeah but they should use same variable (that has to be created as well). The idea is to retract same amount that'll be pushed before purge

closed 2024-09-21 by github-actions[bot]

rongith, 2024-10-17:
To the OP: I created a PR for that: https://github.com/Frix-x/klippain/pull/658
It seems we can't reopen an old issue anymore.