Frix-x/klippain #599: exhaust_filter is running when it shouldn't

kgalilio, 2024-06-06:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-11-g4ff6f5b4

### Describe the bug and expected behavior

The exhaust_filter is supposed to regulate the chamber temperature to a target temperature and/or filter the exhaust air after a print has finished. In my case, as soon as I sat the bed temperature to 100 degrees, the exhaust fan started working, and I had no control over it. I tried bed and chamber soak too, but I kept getting the same result. This prolongs the heatsoak duration.

Surion79, 2024-06-21:
@24naret please create a Feature Request instead of hijacking a similar sounding issue. You can also create an own PR with your idea :)

Surion79, 2024-06-21:
@kgalilio the exhaust fan is a filter fan which is used to have a low underpressure in the chamber so that the fumes are going through the filter instead of siphoning through other holes etc. therefore it must run all the time.  It is not a temperature regulator. If you need a chamber heating regulator, you would need something like ellis bedfan or multipurpose the nevermore filter, which have an innnerchamber circulation air flow.

closed 2024-08-06 by github-actions[bot]