Frix-x/klippain #600: Support for cartographer3d probe

joergenlarsenwm, 2024-06-13:
### Describe the feature or hardware support you'd like

I'd love to see support for the cartographer3d probe.

### Additional context or information

https://docs.cartographer3d.com/cartographer-probe/installation-and-setup/v2-and-v3-probes-w-input-shaping/cartographer-with-input-shaping-v2-and-v3-hybrid

m3cinc, 2024-06-13:
idem here. This probe is actually better than the cnc tap...

Surion79, 2024-06-19:
well, actually, cartographer is a copied design of the beacon probe. 
cnc tap is a complete different probe approach.

shaxs, 2024-09-10:
I wanted to check and see how this was coming along? I would love to see the Cartographer Survey (beacon tap) functionality implemented.

typetetris, 2025-01-08:
I made a pull request of what I am using: https://github.com/Frix-x/klippain/pull/684
Maybe its of use for others, too?

mjonuschat, 2026-06-13:
Implemented in the Z probe framework, confirmed working after 6ea2917db10957f2068b6ac7b7b133ef2c05d36d.

Closing this out as resolved.