Frix-x/klippain #605: SAVE_CONFIG section 'probe' option 'z_offset' conflicts with included value

smweil, 2024-06-19:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-14-gf01c4ca1

### Describe the bug and expected behavior

When running probe_calibrate I cannot save the changes to my config. I'm overriding a few probe commands, but not z_offset. I'm using a Voron TAP for reference. 
Thanks!

### Additional information and klippy.log

[klippy-24.log](https://github.com/user-attachments/files/15906502/klippy-24.log)

EricZimmerman, 2024-06-19:
You have to just update the value as an override. Save config does not work with klippain

Surion79, 2024-06-20:
this is a known topic of klipper that using save_config is limited. Due to the nature of the architecture of klippain with many different sets of configuration, makes it impossible to have a correct value being put in printer.cfg.

EricZimmerman, 2024-06-20:
Danger clipper supports save config regardless of where the actual value is but I don't know if it's going to actually update your printer.cfg or if it would expect to replace the actual section which wouldn't work very well for how things work

smweil, 2024-06-21:
Got it, thank you.

closed 2024-06-21 by smweil