Frix-x/klippain #610: CANCEL_PRINT error when a custom filter fan name used

Sparkss, 2024-06-21:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-16-g44f21b25

### Describe the bug and expected behavior

The cancel print macro does a status call against a fixed fan object 'fan_generic filter'. 

cancel_print.cfg
CANCEL_PRINT
    # If a filter is connected, and used during the print, continue filtering the air
    # for a couple of min before stopping everything
    {% if filter_enabled %}
        {% if printer['fan_generic filter'].speed > 0 %}
            {% set FILTER_TIME = params.FILTER_TIME|default(filter_default_time)|int %}
            START_FILTER SPEED=1
            UPDATE_DELAYED_GCODE ID=_STOP_FILTER_DELAYED DURATION={FILTER_TIME}
        {% endif %}
    {% endif %}

However it is possible to have a custom filter fan and not use 'filter' by leveraging the _USER_VARIABLES variable_filter_name. As is supported in the filter macros.

filter.cfg
START_FILTER / STOP_FILTER
    {% set filter_name = printer["gcode_macro _USER_VARIABLES"].filter_name %}
    SET_FAN_SPEED FAN={filter_name} SPEED=0

But if you use a custom filter name and try to cancel a print job you get an error and cannot cancel/stop the print without e-stop or power cycling (pause still works and I used that first before power cycling)

Error thrown
Error evaluating 'gcode_macro CANCEL_PRINT:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'fan_generic filter'

Surion79, 2024-06-21:
please provide the full klippy.log.

Sparkss, 2024-06-21:
I don't have the log handy. I can read the macro code and see the issue, which is why I didn't bother with the log, just the code review and captured error, which pointed to the code section in question. If the issue is truly not obvious please let me know and I will try to elaborate further. I will also see if the klippy log is still available when I get home, which it might not be as that happened yesterday and I had been cleaning up files lately.

Surion79, 2024-06-26:
the non correctly formatted text is really hard to read. do you mean this line?
`{% if printer['fan_generic filter'].speed > 0 %}`

rajenki, 2024-07-16:
I'd like to second this issue. @Surion79 that is indeed the correct line in the cancel and end print macros.

My use case: I use a Nevermore StealthMax with the Nevermore Controller. The fan isn't called filter, which means it throws the same error that @Sparkss mentioned on print end/cancel. If there's a way to specify the name to be used in these macros, I imagine both of these issues would be solved.

closed 2024-08-30 by github-actions[bot]