Frix-x/klippain #612: Support for dockable probes with fixed z-height

jannwieland, 2024-06-22:
### Describe the feature or hardware support you'd like

I am using a dockable probe at a fixed z-height. Currently there is no option to for the _ATTACH_PROBE and _DOCK_PROBE macros to use a specific z-height.

[image]

In the current state, the macros fail if `printer.toolhead.position.z > probe_min_z_travel` since the toolhead misses the dock.

### Additional context or information

### Possible Solution:

Add a user variable like `probe_z_fixed` and change the `probe_min_z_travel` check in both _ATTACH_PROBE and _DOCK_PROBE to something like this:

```cfg
# Move to safe Z
{% if (printer.toolhead.position.z < probe_min_z_travel or probe_z_fixed) %}
    {% if verbose %}
        { action_respond_info("Moving to a safe Z distance") }
    {% endif %}
    G0 Z{probe_min_z_travel} F{z_drop_speed}
{% endif %}
```

Surion79, 2024-06-26:
question: how does the z probe work in this case when you have unhomed Z? do you use another z endstop? do i see it on the right?

jannwieland, 2024-06-28:
Sorry for the late reply.
Yes it's on the right. I use the standard z-endstop from the Voron 2.4 kit for homing and the klipper_z_calibration. 

I've recently added a [blobifier](https://github.com/Dendrowen/Blobifier) to my setup and had to relocate the klicky probe.

closed 2024-09-27 by github-actions[bot]