Frix-x/klippain #613: Klippain shake&tune broken in klippain config

joergenlarsenwm, 2024-06-25:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-17-gdf5ad5bf

### Describe the bug and expected behavior

not able to enable klippain shake and tune. 
config/software/shaketune.cfg references missing directory and file:

[gcode_shell_command shaketune]
**command: ~/printer_data/config/scripts/K-ShakeTune/shaketune.sh**
timeout: 600.0
verbose: True

Neither the K-ShakeTune nor the shaketune.sh is present on my system.

Checking through the github repos for Shake&tune and klippain and cannot find it.

### Additional information and klippy.log

shell_command: Command {shaketune} failed
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_shell_command.py", line 53, in cmd_RUN_SHELL_COMMAND
    proc = subprocess.Popen(
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1024, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.11/subprocess.py", line 1901, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/printer_data/config/scripts/K-ShakeTune/shaketune.sh'
Error running command {shaketune}

EricZimmerman, 2024-06-25:
Remove all that and start with the installer and v4.

joergenlarsenwm, 2024-06-25:
It seems I was stuck between old and new shake&tune.
Reinstall and adding shaketune tag in printer.cfg seems to work.

closed 2024-06-25 by joergenlarsenwm