Frix-x/klippain #618: Documentation for start print custom module is unclear

Zanonia, 2024-07-02:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-17-gdf5ad5bf 

### Describe the bug and expected behavior

I have followed the instructions in the start print macro  to use the custom start print module but the code is not taken into account by klipper.

For exemple have had this code in my overrides.cfg :

```
[gcode_macro _MODULE_CUSTOM1]
gcode:
    # Load skew profile
    SKEW_PROFILE LOAD=my_skew_profile 
```

And the start sequence stay the same as before, ignoring the module.

sima-it, 2024-07-04:
I have the save problem, I'm running a Nevermore Stealthmax and I'm trying to start the filter in the Start Macro:

`[gcode_macro _MODULE_CUSTOM1]
gcode:
    NEVERMORE_PRINT_START`

Surion79, 2024-07-14:
you need to add the custdom modules in the last of startprint_actions in variables.cfg to be recognized/executed

sima-it, 2024-07-15:
This is the last section in my overrides.cfg
```
[gcode_macro _MODULE_CUSTOM1]
gcode:
    NEVERMORE_PRINT_START
``` 

its not working!

Zanonia, 2024-08-04:
> you need to add the custdom modules in the last of startprint_actions in variables.cfg to be recognized/executed

I can't find "startprint_actions" in variables.cfg file. Is there an example somewhere?

LuisWinter, 2024-08-31:
> you need to add the custdom modules in the last of startprint_actions in variables.cfg to be recognized/executed

is there an explanation or example of it anywhere?

zanmoskotevc, 2024-09-26:
@LuisWinter read about start print sequence here:

https://github.com/Frix-x/klippain/blob/main/user_templates/overrides.cfg

Zanonia, 2024-10-29:
Finally I have edited  manually the start macro. Is there a way to deactivate the tracking of changes and the update from the git to manage again the config manually while keeping the current files?