Frix-x/klippain #625: Critical error. I have a problem with my Octopus pro v1.1 board. I think one of the pins is misidentified.

Egorka555, 2024-07-22:
### Klippain branch

- [X] I confirm using the main branch

### Version

Latest version

### Describe the bug and expected behavior

I have a problem with my Btt Octopus pro v1.1 board. I think one of the pins is misidentified. 
In the config/mcu_definitions/main/BTT_Octopus_Pro_v1.1.cfg , pin MCU_MOTOR3_ENABLE is listed as PA2, but there should be PA0.
In the config/mcu_definitions/main/BTT_Octopus_Pro_v1.1.cfg , pin MCU_HE0 is listed as PA0, but there should be PA2.
"As a result of this error, power is being supplied to the hotend. It is heating up uncontrollably. Mine heated up to 370°C before I noticed the problem. Please fix this as soon as possible.

![1](https://github.com/user-attachments/assets
[image]
[image]
[image]
/e120dfd7-279c-4d75-9c9c-969291c1251c)

Egorka555, 2024-07-22:
[image]

stablestud, 2024-07-22:
I can confirm this. BTT had changed the pins between v1.0 and v1.1

See the README.md of this official board repo:
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/tree/master
It includes the changed PINs in the changelog

stablestud, 2024-07-22:
Hmm, I was not right.
I looked further into this and Klippain's PINs seem to be correct. (verified with changelogs from my link)

Are you sure you have the Octopus Pro v1.1 and not the Octopus Pro v1.0.1?

Even the official pinout states that HE0 should be PA0 not PA2
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro/blob/master/Hardware/BIGTREETECH%20Octopus%20Pro%20V1.1-Pin.jpg

Egorka555, 2024-07-23:
Based on the pin names, it looks like I have a version 1.0 board. I checked my orders, and I did indeed order an Octopus 1.1 F446. Now I just need to visually check which board is installed in the printer. I don't want to disassemble it right now, but in about 2 weeks, when I will be installing EBB CAN on my printer, I will check and inform you of the results.

MasterOfGamerino, 2024-08-21:
> Based on the pin names, it looks like I have a version 1.0 board. I checked my orders, and I did indeed order an Octopus 1.1 F446. Now I just need to visually check which board is installed in the [...]

The Octopus exists in 4 PCB variants, v1.0, v1.1, Pro v1.0 and Pro v1.1. The specific STM32 MCU on either of them can vary, however this doesn't change anything about the pin names.

The first three PCB layouts share the same pinout, with only the Pro 1.1 being different, as can be read [https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro?tab=readme-ov-file#firmware](here).

If you've got an Octopus 1.1, you chose the incorrect config file.

For me the Octopus Pro 1.1 with an STM32H723 runs flawless with the klippain stock mcu definition for this board.

Egorka555, 2024-08-21:
[image]

MasterOfGamerino, 2024-08-22:
> 
> [image]
> 
> 

Yes, this is indeed the board that you said the order details specified, an Octopus version1.1.

However, the pin definitions file you chose, and refer to in this issue, is for the Octopus _Pro_ version1.1, which has a different pinout.

To be fair; the list of mcus during the klippain setup isn't very well sorted (or at all, actually), making it hard to select the correct one, especially given the similar names.

Since the amount of different boards to choose from is only going to get larger, it would probably help to change that process to be a multi-prompt one, e.g. first get a list of manufacturers to choose from, then see the list of their boards that are already available, and maybe even seperate different revision of those in a third and final prompt...
Might write an issue/PR for that.

closed 2024-10-06 by github-actions[bot]