Frix-x/klippain #635: Add support for CNC Taps

jackmoxley, 2024-07-30:
### Describe the feature or hardware support you'd like

Choatic Labs CNC tap, can't handle the sensitivity the default tap is given in the config. So I would like a specialised config that meets the manufactorers settings. I have added a PR to do just that  #632

Surion79, 2024-08-07:
You can add this in your override. I have also cnc tap of chaoticslab and it works fine with the klippain default settings.
We can only provide one set of settings via klippain in the config sections. Individual changes must be put in overrides.
I don't see the need to add it.

jackmoxley, 2024-08-11:
I can, I was merely adding the manufacturers recommendations. If you don't want it feel free to close.

closed 2024-09-26 by github-actions[bot]