Frix-x/klippain #636: Add Beacon Contact

pireybr, 2024-08-04:
### Describe the feature or hardware support you'd like

I tried to add beacon contact to klippain but failed. Implementing it will be nice

### Additional context or information

https://docs.beacon3d.com/contact/

crkochan, 2024-08-09:
To add additional detail, Beacon Contact adds their own`safe_z_home` function that does not like `homing_override`.

See: 
* https://docs.beacon3d.com/contact/#52-configuration
* https://docs.beacon3d.com/contact/#82-homing-override

If `homing_override` is defined, it causes an error on Klipper startup.

To work around the problem, I removed `macros/base/homing/homing_override.cfg`, but this obviously makes the local clone dirty.

No automatic updates for me. :disappointed:

grimlokason, 2024-09-18:
+1 for Beacon contact integration

laukejas, 2024-12-02:
Has there been any progress on this?

gueee, 2025-02-02:
is there any update or branch with beacon contact yet?

crkochan, 2025-02-02:
There’s feedback on my PR, I just haven’t gotten around to making the requested changes yet. 

I haven’t forgotten, I’ve just been very busy, and it’s probably going to be another couple weeks before I have any cycles to spare.

gueee, 2025-02-02:
> There’s feedback on my PR, I just haven’t gotten around to making the requested changes yet.

I read that feedback, would it help if I jumped in implementing the requested changes?

crkochan, 2025-02-02:
By all means.