Frix-x/klippain #638: Installed wiped printers config

ReaperXD1, 2024-08-09:
### Klippain branch

- [X] I confirm using the main branch

### Version

v0.12.0-274-gd7d9092a

### Describe the bug and expected behavior

Upon installation via SSH, all my configuration files disappeared without a trace. I am beyond frustrated right now and I feel like issues like this should really be taken seriously. This has cost me big time as I had a very complex setup, which has taken almost a year to achieve. I really have no words right now.

EricZimmerman, 2024-08-09:
No backup? Maybe next time. 

I would suggest getting a klippy.log as it will contain your cfg.

ReaperXD1, 2024-08-09:
I have a backup from a month ago, but things have changed since. I am pretty mad, how does a simple install cause this?

EricZimmerman, 2024-08-09:
It should have backed them up to klippain backup in home folder. 

Always have backups. Use my GitHub backup writeup and avoid this next time. 

What if your SD card died??

ReaperXD1, 2024-08-09:
SD card is okay, just checked via WINSCP and my files are there thankfully. I'm going to do backups through KIAUH start fresh. Frustrating. I should setup auto backups, that's a good idea. Never occured to me. This is still very inconveniencing. Would a Klippy log still be relevant?

EricZimmerman, 2024-08-09:
Doesn't matter now. You have your configs. Get them OFF the pi. Use my GitHub writeup.

closed 2024-09-24 by github-actions[bot]