Frix-x/klippain #661: Incorrect Sense resistor when using BTT Kraken board. !Dangerous!

Svennixx, 2024-10-20:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1

### Describe the bug and expected behavior

When using the BTT Kraken board you have to use the TMC5160 klipper settings even though the drivers are TMC2160. But if you do that you will have to override sense resistor. The sense resistor of the TMC5160 in Klippain is 0.075 but the TMC2160 drivers need it set to 0.022.

If someone does not realises this the actual run_current of drivers will be **_3.4 times higher_**! 

I would suggest adding a separate TMC2160 or BTT Kraken TMC config and include with the correct sense resistor.

### Additional information and klippy.log

Klippain TMC5160:
[image]
Official BTT Kraken settings:
[image]

Nathan22211, 2025-07-10:
I feel like setting a sense resister for 5160s PERIOD is an issue. As the sense resisters on the Mellow 5160s and MKS 5160s are slightly different

mjonuschat, 2026-05-14:
Thanks for reporting this. The safety concern is valid, but Klippain can’t safely solve this with the current Klipper config model.

`sense_resistor` is not just a driver-type setting. It depends on the physical board revision and the driver slot. On the Kraken, BTT documents different values for S1-S4 vs S5-S8, and those values also changed across board revisions. Klippain’s TMC includes only know the logical axis/stepper, not which physical Kraken slot it is wired to, so a generic TMC2160 or Kraken-specific include could silently set the wrong value.

Removing the value is not safe either, because Klipper falls back to its own default instead of forcing an explicit choice.

For now, this has to stay an explicit per-printer override in user config, based on the exact Kraken revision and physical slots in use. I’m closing this as a known limitation instead of trying to add workarounds that would ultimately have the risk of producing different unsafe configurations.