Frix-x/klippain #681: Problem configuring extruder on Fly-SB2040 Pro

bertenvdb, 2024-11-21:
### Klippain branch

- [X] I confirm using the main branch

### Version

v4.3.1-18-ga30308e5

### Describe the bug and expected behavior

Spider v3 + Fly-SB2040 Pro Max

I haven't looked at my Voron for several months but now have some spare time to continue working at it. Sometime last summer I installed the Fly-SB2040 Pro, and everything worked perfectly (without Klippain). However, I discover Klippain and I thought it's clean modular config was worth installing, so I did. 
I've followed the relevant documentation, but can't get my extruder to work. Everything else on my toolhead works, except for the extruder. Whatever I do, there's always some error. I've been going at this for about 8 evenings now, and desperately need help. 

With the config as I think it should be, I get: `Unable to read tmc uart 'extruder' register IFCNT`

Please help.

### Additional information and klippy.log

[klippy.log](https://github.com/user-attachments/files/17852654/klippy.log)
[printer.txt](https://github.com/user-attachments/files/17852461/printer.txt)
[overrides.txt](https://github.com/user-attachments/files/17852462/overrides.txt)
[mcu.txt](https://github.com/user-attachments/files/17852463/mcu.txt)

bertenvdb, 2024-11-23:
I've noticed config/hardware/mcu_definitions/toolhead/Mellow_SB2040_Pro.cfg contains this definition:
`MCU_EMOT_CS=gpio8`

But according to the [pinout](https://raw.githubusercontent.com/Mellow-3D/klipper-docs/refs/heads/master/docs/images/boards/fly_sb2040_pro/pinout.jpg) this should be gpio11.

Changing this does not result in solving the problem, Klipper still says `Unable to read tmc uart 'extruder' register IFCNT`.

Frix-x, 2025-01-08:
I'm coming back on the topic since a some time on Shake&Tune...
Did you solved your issue? Usually the error with IFCNT is indeed a bad wiring or bad configuration of the pins...