Frix-x/klippain #692: Uninstalling Shake&Tune removes all config files

dazza2924, 2025-01-22:
### Klippain branch

- [x] I confirm using the main branch

### Version

Cannot tell - because its deleted everything. It was v5 that I at least remember

### Describe the bug and expected behavior

There seems to be a massive bug in when you use the command on the readme (https://github.com/Frix-x/klippain/blob/main/README.md) to uninstall klippain, it just literally deletes your entire klipper/mainsail config, and renders the entire printer useless. Luckily, I saw someone else mention on this github that they had the same issue so I saved what was in my config folder, but this is really bad.

NanoMetere, 2025-01-28:
We have no Uninstall script on Klippain Shake&Tune now.
and this script you mentioned is for Klippain uninstall not Klippain Shake&Tune.

maybe we should add if Klippain is installed in pre-flight checks

A6uh, 2025-02-12:
I would highly suggest adding those pre-flight checks or at the very least a disclaimer. I was also trying to uninstall Shake&Tune (it kept crashing my pi 3B+) and the script here deleted all of my config files. Unfortunately, I didn't see this like OP did, and didn't backup all of my files beforehand. All of my configs were very finely tuned, so I'm pretty frustrated with it.

MrSalzi, 2025-02-28:
The same here. As known shaketune never worked and now my printer is broken. Thank you.

ohyeah8000, 2025-03-04:
Oh oh!

I executed the uninstall command and also here all my config files are gone. 

wget -O - https://raw.githubusercontent.com/Frix-x/klippain/main/install.sh | bash

Funny and didn't know about this before. 
Will nuke my config now and install everything from scratch. No big deal. 

However I'd like to understand what happened and what the problem was. Does anyone know?

Felix, great work. Klippain is awesome.