Frix-x/klippain #699: Backup function

iSayhello, 2025-02-16:
### Klippain branch

- [x] I confirm using the main branch

### Version

latest

### Describe the bug and expected behavior

Thank you for the helpful backup function of klippain when installing. Now its not just bread as you say. 

[image]

the-murph, 2025-02-20:
Same thing just happened to me. Files are missing from the backup, i also noticed that Klippain didn't backup any directories in my config folder so those are gone too.

Karl-Barx, 2025-03-05:
Same thing is happening to me. Also leaves everything as read only before this happens. Gave up and went back to my homebrewed cfg