Frix-x/klippain #708: Kalico support planned?

RodentVienna, 2025-03-05:
### Describe the feature or hardware support you'd like

Hello,

Kalico (previously DangerKlipper) also has some advanced, not-so-dangerous features.
Do you have any interest to look at integrating this (optionally) in the future?

Or is the genral consensus, that Kalico is only for advanced users, that do not base on Klipper-Suites such as Klippain, or DemonKlipperEssentials?

Thank you fou your opinion on this.
Best, Alex

stokedgsxr, 2025-05-13:
I'm curious about this as well

mUbsta, 2025-05-13:
I've been running with Kalico for quite some time now (definitely before the name change) without any issues or workarounds at all!  I've obviously not using every feature in Kalico or Klippain but I've had enough experience to say there's no harm in trying.  You can roll it back pretty easy if it breaks!

NanoMetere, 2025-08-02:
Using 2 machines(2.4 and 0.2) running klippain and kalico without any issues.
can use almost every kalico additions with override.cfg if you want. 

im using additions listed below.
- - -
## Model Predictive Control ( V0.2 )
```
[extruder]
control: mpc 
heater_power: 100  # 50w * 2
cooling_fan: fan
filament_diameter: 1.75
filament_density: 1.06
filament_heat_capacity: 1.8 
step_pulse_duration: 0.000004
```
- - -
## Drop first probe result ( V2.4 Voron Tap )
```
[probe]
drop_first_result: True 
```
- - -
## Non Critical MCUs
### V0 adxl
```
[mcu adxl]
serial: /dev/serial/by-id/usb-Klipper_rp2040_E6635C08CB718927-if00
is_non_critical: True
```
### v2.4 Chamber Heater
```
[mcu tartatus]
canbus_uuid: 61a68c03f57b
is_non_critical: True
```
- - -
## Adaptive QGL ( V2.4 )
```
[quad_gantry_level]
min_horizontal_move_z: 2.0
adaptive_horizontal_move_z: True
```
- - -
## Temperature fan - control: curve ( V2.4 TMC5160 Fan)
```
[temperature_fan TMC5160]
control: curve
points:
40.0, 0.0
45.0, 0.1
50.0, 0.2
60.0, 0.4
65.0, 0.7
80.0, 1.0
cooling_hysteresis: 3.0
```
- - -