Frix-x/klippain #713: Cartographer 3D

EuroAfrican, 2025-05-18:
### Describe the feature or hardware support you'd like

Hi There

I just installed Klippain on a Trident AWD with a Cartographer, so it did not work. 

It would be great to have this functionality added into the system. 

thank you,.

Cicatr1x, 2025-05-19:
There is also some work in progress that you could use by now.

https://github.com/Frix-x/klippain/pull/684

Cicatr1x, 2025-07-10:
> wait so... does beacon work or is it a similar siutation?

Cartographer works with the little workaround yeah.

ErdbeermarmeladebrotmitHonig, 2026-01-07:
Hi,
I opened a pull request here: #743 
This also uses the newer macros comming with the cartographer firmware.